martim01/pam #8: Trouble cloning repo on Mac OS

njh, 2019-07-16:
Hi,

I have just been taking a look at PAM and thought I would try cloning the git repository.

It had a problem checking out because by default Mac have case-insensitive filesystem and cannot have both a file called `Pam.html` and `pam.html` in the same directory. 

Would this be an easy thing to change?

Thanks,

nick.

```
warning: the following paths have collided (e.g. case-sensitive paths
on a case-insensitive filesystem) and only one from the same
colliding group is in the working tree:

  'documents/help/Pam.html'
  'documents/help/pam.html'
```

martim01, 2019-07-17:
Hi,
Looks like Pam.html was a copy of pam.html. I've deleted Pam.html so you should be able to clone okay now.
I've not had the chance to try compiling PAM on a Mac so, whilst everything used should be cross platform, can't guarantee the results. I'd be interested to know how you get on if you attempt this.

Matt

closed 2019-07-17 by martim01

njh, 2019-07-17:
That fixed it, thanks!