martim01/pam #9: Trying to run on Raspberry Pi: Segmentation fault

njh, 2019-08-11:
Hi,

I recently got a Official Raspberry Pi Touch screen, in order to try and give PAM a go. My goal is to get it working with AES67 and a [Glensound AoIP44](https://www.glensound.co.uk/product-details/aoip44/).

I followed the instructions in the Readme and this wiki page: https://github.com/martim01/pam/wiki/Turning-a-Raspberry-Pi-into-a-PAM

But have experienced a few problems.

## Setting Up

I started with a fresh copy of Raspbian: `2019-07-10-raspbian-buster.img`

When starting PAM there were some error messages about `bluealsa`. Given that I have no plans for using Bluetooth, I decided to uninstall it, which made the error messages go away.
```
apt remove --purge bluez pi-bluetooth bluealsa
```

I also disabled the Rapberry Pi's built-in audio in preference for a USB Audio card by editing `/boot/config.txt`:
```
dtparam=audio=off
```

I also had an error message about not being able to find a device. I noticed that `eth0` didn't exist, so turned off predictable network interface names:
> sudo raspi-config
> Select "2. Network options"
> Select "N3 Network interface names"
> Select "No". It's now disabled predictable network interface names.
> Click "OK" and "Finish". Then it will reboot your raspberry pi.

I also had a message that was implying that I need to type in a password for sudo. To keep things simple I decided to add the following line to `/etc/sudoers`, instead of the multiple lines for `/sbin/shutdown` etc...
```
pam ALL=(ALL) NOPASSWD:ALL
```

## Current state

Finally I was left with the following messages after running `/usr/bin/pam2`

```
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
SIOCADDRT: File exists
Segmentation fault
```

* I assume that ALSA error messages are harmless?
* Is the `SIOCADDRT` message expected?

I order to work out what is causing the Segmentation fault, I used `gdb` and it looks like it is a problem with calling `SetBackgroundColour`:

```
Thread 1 "pam2" received signal SIGSEGV, Segmentation fault.
0x76437e7c in uiRect::SetBackgroundColour(wxColour const&, wxColour const&) () from /home/njh/pam/pam2/../lib/libpambase.so
(gdb) bt
#0  0x76437e7c in uiRect::SetBackgroundColour(wxColour const&, wxColour const&) () from /home/njh/pam/pam2/../lib/libpambase.so
#1  0x76469664 in wmSlider::SetButtonColour(wxColour const&) () from /home/njh/pam/pam2/../lib/libpambase.so
#2  0x0016e188 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

I would be very grateful if you had any suggestions on how to fix this, or amend my setup.

Thanks!

nick.

martim01, 2019-08-14:
HI Nick,
I've not had a chance to try setting up PAM on buster yet only jessie. I guess this is what has caused quite a few of your issues.
The Alsa errors are indeed harmless.
I'll have a look at what I'm doing in the SetBackgroundColour part of the code and get an image of buster to try it out on.
Matt

martim01, 2019-08-19:
I've downloaded Buster  and made some changes to the code (mainly changing virtual functions that shouldn't have been to non-virtual and making sure class members were correctly initialised in the constuctor). I've built the resulting code using CMake and Codeblocks on Buster and Stretch and am no longer getting Segmentation faults.

closed 2019-08-19 by martim01

njh, 2019-08-19:
Fantastic!

I will give the updates a go.

njh, 2019-08-21:
Successfully compiled and installed. Thanks!

[image]