martim01/pam #12: soundcard

dewiweb, 2020-01-07:
Is there a solution to use the onboard audio out (mini-jack) in place of a specific pi-Hat?(or maybe also an usb sound card)

martim01, 2020-01-07:
If you are building/installing the code yourself then the on board audio out and any compatible usb sound card should be available. PAM uses portaudio which itself uses asio, so any asio compatible in/out should be discovered.

If you are using the prebuilt image then the mini-jack output has been disabled. You need to edit /boot/config.txt to reenable it.

dewiweb, 2020-01-08:
It could be nice to have the ability to control output volume from pam interface.
(I'm using alsamixer for now but I've to alt+tab between pam and terminal to adjust it....)

martim01, 2020-08-14:
Latest version allows gain control of outgoing audio if using Soundcard as output

closed 2020-08-14 by martim01