martim01/pam #13: pam user password in prebuilt image

dewiweb, 2020-01-08:
Hello Martin,
What's the password for pam user in prebuilt image.
I'm looking for edit /boot/config.txt file...

martim01, 2020-01-08:
I think in order to edit /boot/config.txt you'll need super user rights which the pam user deliberately doesn't.
Try 
`su pi `
with password **raspberry**

dewiweb, 2020-01-08:
Thanks for reply,
It works !

closed 2020-01-09 by martim01

Gemini2350, 2020-07-18:
Same problem here with the newest Image. But with 
"su pi"  and password raspberry i get 
"su: Authentication failure"
Can you help me? love your tool but i also need to turn the screen ;)

martim01, 2020-07-18:
Sorry - looks like I changed the password between releases.
The password for  user **pam** is **3045bh** but the user is quite locked down.
The password for user **pi** which allows sudo is **vuByqU43cdRSCHkp**

Fischmaster3000, 2022-06-04:
did you change the password again? can't login into the latest prebuild image :(

martim01, 2022-06-06:
Did you not get the setup application when you first booted the new image? This should give you the opportunity to set the password to what you want.

It's possible that the new image has the default password of "raspberry" - the idea being you'd change it with the setup app.

Can you let me know whether the setup application started, and if "raspberry" works. If not then I'll burn a copy of the image for myself and workout what I set the password to.