martim01/pam #14: Good job Dear

yboujraf, 2020-02-22:
Dear @martim01 

I 'd like to say : Very good job for this project.

Don't stop ;-)

Remarks : do you think the same dev in Typescript is possible or limited because all the library are C++ ?

Best Regards,
Youssef

martim01, 2020-02-24:
Hi - Thanks for the kind comments, glad you like the project.
I'm afraid I don't know enough about Typescript to say whether it would be possible. There are libraries such as nbind which allow you to access C++ libraries from Javascript and Typescript so this might be the way to go...
Matt

yboujraf, 2020-02-28:
Dear @martim01 

Do you mean : https://github.com/charto/nbind to convert C++ to Typescript ?
I will try it.

# AoiP:

## NMOS

- which library are you using for nmos ? Is it Sony ?

## Dante

Did you have tried to control Dante routing ?

## Ravenna AES67

Did you have tried to control Ravenna routing ?

# PTP

Could you read or extract LTC through PTP ?

Best Regards
Youssef

martim01, 2020-02-28:
Hi Youssef,

nbind was the project I'd seen - but as I say I've little experience of Typescript.

# AoIP:
## NMOS
The [library] (https://github.com/martim01/nmos) I use is one that I wrote from scratch myself - it implements a IS-04/05 node fully. I'd looked at the Sony project but at the time it needed Bonjour rather than using Avahi and I couldn't figure out how to make it work in a practical situation. This was a quite a while ago and I know it is a much more mature project now (and covers many more of the IS-xx than my library).

## Dante
I had a look at implementing Dante routing, but I can't decode a Dante audio stream unless it is in AES67 compatibility mode. If it is in AES67 compatibility mode then the Dante streams are advertised using SAP and the PAM will discover them. It is then possible to select them on the PAM to monitor.

## Ravenna AES67
If you mean  - mDNS-SD to advertise the Ravenna sources and RTSP to then initiate the RTP sessio then yes, PAM implements all of this.

## PTP
PAM can decode and generate LTC. It can also act as a PTP slave this is still a work in progress) so it would be possible to sync the LTC that is generated by the PAM to a PTP master clock. I haven't yet implemented this though.

yboujraf, 2020-02-28:
Dear @martim01 

## AoIP:

### NMOS
The [library] (https://github.com/martim01/nmos) 

- Do you think to continue to update your library ? instead of using other libraries maybe (more contributors) but when I see you are alone and in a minimum time you implement IS-04 & 05, IS-08 could be done also (audio mapping) to control the monitoring L to Right, Right to Left, Both, ....

What do you think ?

### Dante

I have for windows, a license to share if you want then you could verify easily.

### Ravenna AES67

That's sound good. I need to try it soon.

## PTP
PAM can decode and generate LTC. It can also act as a PTP slave this is still a work in progress) so it would be possible to sync the LTC that is generated by the PAM to a PTP master clock. I haven't yet implemented this though.

=> How to generate the LTC from PTP, what do you mean ? on network, or analog through audio card or both ? I am interesting
=> Could we able to decode analog LTC and encoded in the PTP as well ? for example I have a Old VTR (when playing I want to use the LTC from the tape)

## Private

Finally, do you have a Private email ?

Best Regards
Youssef

martim01, 2020-03-02:
## NMOS
I hope to update my library - I would like to get the IS04/05 client section complete and also IS08, it's a matter of fitting it around other more pressing things though.

## Dante
Thanks for the offer. I actually have a FocusRite RedNet unit that I have been experimenting with. If you put the unit in AES67 mode then it easy to subscribe to audio streams from the PAM using SAP.  I'm not intending to implement a Dante specific mode as their transport mechanism is proprietry and closed-source.

## PTP
PAM has a built in LTC generator. Depending on the "soundcard" connected to the unit and chosen output device this LTC stream can be sent out as Analog, Digital or AES67.

PAM will decode an Analog LTC audio stream, however my PTP code currently only acts as a PTP slave not as a Grandmaster. 

## Private
I've added my email address to my profile.

yboujraf, 2020-03-02:
## NMOS

I wish you finalize the implementation IS04/05/08

## Dante

I am interested ONLY for AES67.
I will check with Dante infrastructure if it is also setted as AES67 instead of Dante Specific.

## PTP

I will try to forward Analog LTC from AES67

## Private

Thanks for the email.

Best Regards
Youssef

closed 2021-12-04 by martim01