martim01/pam #30: Feature request: Simper user interface for operation, workflow driven.

riban-bw, 2020-03-06:
The user interface provides access to operational and configuration elements. Some user workflows require access to both. This can be confusing for operators. It would be good to have simplified user interface option and / or hide all configuration behind one button.

Most significantly, the selection of sources, particularly AoIP streams should be available to the operator more easily. It shouldn't necessarily be part of the configuration settings and should be part of the user workflow. Maybe configuration settings could allow each source to be enabled and then a separate user workflow based selection of input be provided.

The current workflow to select AoIP source is:
- Select the view, e.g. Screens->Meters
- Touch unused part of screen to show menu buttons
- Select "Settings"
- Select AoIP (may already be selected)
- Select AoIP stream
- Select "Screens"

This is a lot of touches (4) to swap between sources. The user must switch context between view selection and settings adjustment which isn't intuitive.

martim01, 2020-03-06:
Latest code has press "AoIP" button on bottom right to bring up screen of AoIP sources. Selecting one will return to previous screen

martim01, 2020-03-06:
Like the idea of filtering the list in the Settings before presenting to the new screen. I'll have a think about how to do that