martim01/pam #43: wxWidgets with vs code?

yboujraf, 2020-03-13:
Dear @martim01 

I am using vs code on mac and Windows, did you tried one of them or still using as recommenced in the repository?

Best regards
Youssef

martim01, 2020-03-14:
I haven't tried compiling PAM with vs code but I think I'd should work fine.
wxWidgets provides a visual studio project for building their library. You'd need to make your own project/solution for the rest of the PAM libraries and applications though. 
I think you can still use GCC as your compiler as well if you wish.

If you get it working it would be good to know and add as an option for future users.

Matt

yboujraf, 2020-03-20:
I am back, I am using windows 10.
And would like to compile your project.
Do I need to install GCC for windows but I think it is 32 bits not 64?

Best Regards
Youssef

martim01, 2020-03-20:
Hi,
Probably the easiest way to compile on Windows 10 is to download CodeBlocks (this includes the MinGW compiler etc). You can then open up the relevant projects/workspsaces in Codeblocks and all should compile fine.
I quite often do bits of PAM development on Windows 10 and this is the method I use.

Matt

yboujraf, 2020-03-20:
Regarding mingw compiler i downloaded but no exécutable,  do.i.need to copy
on a dedicated folder ?

martim01, 2020-03-20:
The executables should live in MinGW/bin. 
As I say I use the MinGW build that comes as standard with Code::Blocks, but I'm pretty sure the MinGW standalone download puts everything in that folder as well

yboujraf, 2020-03-20:
Ok thanks

closed 2021-12-04 by martim01