martim01/pam #48: Cmake build error - winsock2.h

riban-bw, 2020-03-19:
Building using Cmake on Raspbian throws an error:
```
pam/external/ptpmonkey/include/ptpstructs.h:7:10: fatal error: winsock2.h: No such file or directory
 #include "winsock2.h"
```
I have tried running cmake twice. Commands used:
```
cmake .. -DoptionPTP=ON
cmake --build .
```

riban-bw, 2020-03-19:
This seems to be because the build process is not asserting the __GNU__ preprocessor directive.

martim01, 2020-03-19:
This bug should have been fixed on the PTPMonkey repository. If PAM has already downloaded the repository (into external/ptpmonkey) you'll need to pull it down again from Github.

closed 2020-03-19 by martim01

riban-bw, 2020-03-19:
I found the ptpmonkey repro under pam/external/ptpmonkey and run `git pull origin master` after which compilation progressed. Does this need anything changing within the build system to avoid future incident or documentation, e.g. README or are you confident it is resolved for all users (other than the unfortunates like me who happened to hit this issue at this time)?