martim01/pam #49: Could not find correct symbols

riban-bw, 2020-03-19:
I have followed the cmake build instructions from README but when I run the resulting compiled binary I see errors and it does not get beyond a splash screen.
Extract from output:
```
...
GetInputDevice: Done
PTPMonkey listening on interface eth0 for domain 0
Executable Path = /root/pam/bin
Core Lib Path = /usr/local/lib/pam2
Plugin Monitor Path = /usr/local/lib/pam2/monitor
Plugin Test Path = /usr/local/lib/pam2/test
Documents Path = /root/pam
Logs Path = /root/pam/logs
Wav Files Path = /root/pam/wav
Temp Path = /tmp
Monitor Lib Directory: /usr/local/lib/pam2/monitor
Lib : spectogram
Lib : aoipinfo
Lib : radar
Lib : peakcount
Could not find correct symbols in lib /usr/local/lib/pam2/monitor/libpeakcount.so
Lib : r128
Lib : LTCGenerator
Could not find correct symbols in lib /usr/local/lib/pam2/monitor/libLTCGenerator.so
Lib : anglemeters
Lib : polarscope
Lib : waveform
Lib : meters
Lib : LTC
Could not find correct symbols in lib /usr/local/lib/pam2/monitor/libLTC.so
Lib : peaklog
Could not find correct symbols in lib /usr/local/lib/pam2/monitor/libpeaklog.so
Lib : fft
Lib : scope
Lib : identify
Could not find correct symbols in lib /usr/local/lib/pam2/monitor/libidentify.so
Lib : lissajou
Lib : record
Could not find correct symbols in lib /usr/local/lib/pam2/monitor/librecord.so
Output source is input
Create Audio Destination Device: Soundcard
Open Audio Device: Soundcard
Pi build currently does not support duplex streams
Open Audio Device: Soundcard
Pi build currently does not support duplex streams
Session Changed: 1
```
This is preceded by alsa messages which I don't think are relevant to this issue and will be reported separately if necessary.

martim01, 2020-03-19:
What happens if you click on the splash screen?

martim01, 2020-03-19:
The plugins that the application is complaining about are not monitor plugins but test plugins, so this is not a bug. What is strange is why they are in that folder or any the application is trying to load them

riban-bw, 2020-03-19:
Sorry - didn't have a working touchscreen / mouse. Clicking on splash screen does open app. As you say though, need to figure out why these plugins are installed / selected.

martim01, 2022-01-19:
This was an issue with where the cmake install was putting the plugin libraries. This is now fixed

closed 2022-01-19 by martim01