martim01/pam #53: Feature: Customisable SAP announcement group address

PheedbackPhil, 2020-04-16:
The group address for listening for SAP announcements apears to be hard coded to 239.255.255.255, whereas commonly used programs like FFMPEG and VLC use  (I think) 224.2.127.254 - would be nice to have this configurable as this had me confused for a while!

martim01, 2020-04-16:
I'd hardcoded it to what Dante in AES67 mode used. Hadn't occurred to me that other applications might use a different address. I'll make this configurable next time I make some changes

PheedbackPhil, 2020-04-16:
Ah I wondered if it was something like that! Was very happy to see PAM being used in BBC CSq for audio monitoring, by the way!

PheedbackPhil, 2020-04-16:
You are absolutely right of course, @njh - I note that when I run VLC on my OSX box I see it listens on all three group addresses...

```
IPv4 Multicast Group Memberships
Group               	Link-layer Address	Netif
239.255.255.255     	1:0:5e:7f:ff:ff 	en0
239.195.255.255     	1:0:5e:43:ff:ff 	en0
224.2.127.254       	1:0:5e:2:7f:fe  	en0
```
I guess it would be nice for PAM to do the same perhaps?

martim01, 2020-04-16:
Sounds sensible, and should be simple enough to do

martim01, 2020-04-21:
I've altered the discovery panel to give the user the option to monitor local, organistation and/or global SAP rather than just local.
I'm currently not able to test that this works fully.