martim01/pam #55: github - clone issue

yboujraf, 2020-07-23:
Dear @martim01 

I tried to clone the repo and I got this error message :

```
desk01@BY_DESKTOP_04 MINGW64 /c/Github
$ git clone https://github.com/martim01/pam.git
Cloning into 'pam'...
remote: Enumerating objects: 365, done.
remote: Counting objects: 100% (365/365), done.
remote: Compressing objects: 100% (234/234), done.
error: filename in tree entry contains backslash: 'C:\taps.txt'
fatal: Not all child objects of 194234ace4d5f1b0222d0b1c465cc6e339a75032 are reachable
fatal: index-pack failed
```

Thanks if you could fix it.

Best Regards
Youssef

martim01, 2020-07-24:
Hi,
I can't reproduce this error on either a Windows or Linux machine. I also see 368 objects rather than 365. Can you let me know what OS you are using and which branch your are attempting to clone?

njh, 2020-07-24:
I was successfully able to clone the repo on Linux but it is currently failing on Mac:

```
warning: the following paths have collided (e.g. case-sensitive paths
on a case-insensitive filesystem) and only one from the same
colliding group is in the working tree:

  'external/live/testProgs/sapWatch.cpp'
  'external/live/testProgs/sapwatch.cpp'
```

(Mac OS has a case-insensitive filesystem by default)

njh, 2020-07-24:
This may be the offending file though: 

https://github.com/martim01/pam/commit/46f7e23146bb961f1aac5538ec59f4cd08110bd5

martim01, 2020-07-24:
Thanks njh,
I've removed sapwatch.cpp - it was exactly the same as sapWatch.cpp.
Well done on finding the file - your Git expertise exceeds mine! 
I can't see why that should be causing an issue with cloning now though.

yboujraf, 2020-07-24:
Dear @martim01 , @njh 

I am using windows 10 OS and the issue stills

[image]

The error is 👍 
```
error: filename in tree entry contains backslash: 'C:\taps.txt'
```

I hope it could help.

Best Regards,
Youssef

yboujraf, 2020-07-28:
Dear @martim01  , @njh 

On windows 10,

The issue when cloning the repo through GITHUB DESKTOP stills.

The error message : 

``` error: filename in tree entry contains backslash: 'C:\taps.txt'
```

Best regards,
Youssef

njh, 2020-07-29:
@yboujraf Sorry, I don't have Windows.

The repo is now successfully cloning on my Mac again.

One alternative would be to download the repo as a Zip file:
* Latest release: https://github.com/martim01/pam/archive/1.0.1341.7444.zip
* Current state of the `master` branch: https://github.com/martim01/pam/archive/master.zip

yboujraf, 2020-07-29:
Dear @njh 

Thanks for the link.
Yes when downloading the zip file, I don't have any issue but with git on windows, as understood, there is a backslash in a filename.

Best Regards,
Youssef

martim01, 2020-07-30:
I use Tortoise on Windows and haven't seen an issue, so it might be worth trying that. I'm not near a computer this week but will have a look at GitHub desktop when I get back

yboujraf, 2020-08-03:
Dear,

even using the git clone command I have the same issue on windows.

Same error message :

```
error: filename in tree entry contains backslash: 'C:\taps.txt'
```

Best Regards,
Youssef

martim01, 2020-08-04:
It looks like this is an issue with the 2.5.3 version of GitHub Desktop (it's a bug/feature that has appeared in previous versions as well). If you download the beta version of GitHub desktop https://github.com/desktop/desktop#beta-channel then it all works fine.

closed 2020-08-04 by martim01