martim01/pam #59: asio.hpp missing from sapserver/src/parser.cpp

ds1k99, 2021-06-29:
parser.cpp references asio.hpp but is not included within the git tree. Can you offer guidance on this? Is it a boost library that is required?

[ 40%] Building CXX object pambase/CMakeFiles/pambase.dir/__/external/sapserver/src/parser.cpp.o
/home/dave/git/pam/external/sapserver/src/parser.cpp:5:10: fatal error: asio.hpp: No such file or directory
    5 | #include "asio.hpp"
      |          ^~~~~~~~~~
compilation terminated.

martim01, 2021-06-29:
Hi,
Asio is included as a Git submodule so you should be able to pull it using

git submodule update --init --recursive

Or you can clone the repository https://github.com/chriskohlhoff/asio

It's the non-boost version of Asio

ds1k99, 2021-06-30:
Marvellous, that's cleared it up thank you.

Forgive my naivety, but I am also missing log.h from the spectogram plugin. your above advise hasn't yielded anything this time. Any advise on this one?

[ 87%] Building CXX object plugins/spectogram/CMakeFiles/spectogram.dir/spectogrammeter.cpp.o
/home/dave/git/pam/plugins/spectogram/spectogrammeter.cpp:10:10: fatal error: log.h: No such file or directory
   10 | #include "log.h"

martim01, 2021-06-30:
Sorry - that's an error on my part. The include directory was missing from CMakeLists.txt for the Spectogram plugin. If you do a pull on the repository it should now be there.

closed 2021-08-31 by martim01