martim01/pam #63: cmake build error with NMOS

Twister10000, 2021-12-02:
Hello,

I always get the error that the NMOS option was not used. 
`
pi@raspberrypi:~/git/pam/build $ cmake .. -DoptionNMOS=ON
-- Configure pambase
/home/pi/git/pam/pambase
Found pml::log
Found pml::dnssd
Found sapserver
Found ptpmonkey
/home/pi/git/pam/pambase
-- Found avahi
-- Found portaudio
-- Found sndfile
-- Configure pamfft
-- Configure pamlevel
-- Found libsamplerate
-- Configure plugin anglemeters
-- Configure plugin aoip
-- Configure plugin channeldelay
-- Configure plugin distortion
-- Configure plugin fft
STATUS,Build Type: Release
-- Configure plugin fftphase
STATUS,Build Type: Release
-- Configure plugin identify
-- Configure plugin levels
-- Configure plugin lineup
-- Configure plugin lissajou
-- Configure plugin LTC
-- Configure plugin LTCGenerator
-- Configure plugin meters
-- Configure plugin peakcount
-- Configure plugin peaklog
-- Configure plugin polarscope
-- Configure plugin r128
-- Configure plugin radar
-- Configure plugin record
-- Configure plugin scope
-- Configure plugin spectogram
-- Configure plugin InputAlign
-- Configure plugin waveform
-- Configure pam2
-- Found libcurl
-- Found avahi-client
-- Found portaudio
unable to set CAP_SETFCAP effective capability: Operation not permitted
-- Configure pamupdatemanager
-- Found libcurl
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/git/pam/build
`

Another error is that I always get the following error message during the build:(
short:
`[  2%] Built target groupsock
[ 38%] Built target liveMedia
[ 39%] Built target BasicUsageEnvironment
[ 40%] Built target UsageEnvironment
[ 40%] Building CXX object pambase/CMakeFiles/pambase.dir/aes67client.cpp.o
/home/pi/git/pam/pambase/aes67client.cpp: In constructor ‘AES67Client::AES67Client(UsageEnvironment&, unsigned char, const char*, int, const char*)’:
/home/pi/git/pam/pambase/aes67client.cpp:83:53: error: no matching function for call to ‘Groupsock::Groupsock(UsageEnvironment&, in_addr&, int, int)’
`
FULL LOG:
`pi@raspberrypi:~/git/pam/build $ cmake --build .
[  2%] Built target groupsock
[ 38%] Built target liveMedia
[ 39%] Built target BasicUsageEnvironment
[... 94 lines cut ...]
  760 |     char const* prefix = "sip:";
      |                 ^~~~~~
/home/pi/git/pam/pambase/aes67client.cpp: In static member function ‘static Boolean AES67Client::parseSIPURLUsernamePassword(const char*, char*&, char*&)’:
/home/pi/git/pam/pambase/aes67client.cpp:834:17: warning: unused variable ‘prefix’ [-Wunused-variable]
  834 |     char const* prefix = "sip:";
      |                 ^~~~~~
/home/pi/git/pam/pambase/aes67client.cpp: In member function ‘unsigned int AES67Client::getResponse(char*&, unsigned int)’:
/home/pi/git/pam/pambase/aes67client.cpp:920:25: error: cannot convert ‘sockaddr_in’ to ‘sockaddr_storage&’
  920 |           bytesReadNow, fromAddr);
      |                         ^~~~~~~~
      |                         |
      |                         sockaddr_in
In file included from /home/pi/git/pam/pambase/../external/live/liveMedia/include/RTPInterface.hh:30,
                 from /home/pi/git/pam/pambase/../external/live/liveMedia/include/RTPSink.hh:28,
                 from /home/pi/git/pam/pambase/../external/live/liveMedia/include/RTCP.hh:25,
                 from /home/pi/git/pam/pambase/../external/live/liveMedia/include/MediaSession.hh:52,
                 from /home/pi/git/pam/pambase/aes67client.h:24,
                 from /home/pi/git/pam/pambase/aes67client.cpp:1:
/home/pi/git/pam/pambase/../external/live/groupsock/include/Groupsock.hh:142:34: note:   initializing argument 4 of ‘virtual Boolean Groupsock::handleRead(unsigned char*, unsigned int, unsigned int&, sockaddr_storage&)’
  142 |         struct sockaddr_storage& fromAddressAndPort);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
gmake[2]: *** [pambase/CMakeFiles/pambase.dir/build.make:95: pambase/CMakeFiles/pambase.dir/aes67client.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:691: pambase/CMakeFiles/pambase.dir/all] Error 2
gmake: *** [Makefile:149: all] Error 2
`

martim01, 2021-12-02:
### Error: "unable to set CAP_SETFCAP effective capability: Operation not permitted"
In order to allow the PTP library access the requisite ports and set the system clock the setcap function is called on the application once it is built. This function must be called as superuser. I've moved the code so hopefully now it is only called if you run `sudo cmake --build . --target install`. The error can be ignored anyway as it shouldn't stop the code from compiling

### Error: no matching function for call to ‘Groupsock::Groupsock(UsageEnvironment&, in_addr&, int, int)
This was an error with the CMakeLists.txt. The file **aes67client.cpp** should no longer be included in the build. I've now deleted them and removed the reference to it in CMakeLists.txt

### NMOS build
I believe the NMOS build should currently work but is not complete. I have been doing a fair amount of rewriting the NMOS library and PAM in order to make them work properly. I'm hoping to merge this work with the main branches shortly.

Twister10000, 2021-12-03:
The build is now working. 
Unfortunately I still get the following error message when I try to execute this command:
sudo cmake --build . --target install
[  1%] Built target UsageEnvironment
[  3%] Built target groupsock
[ 38%] Built target liveMedia
[ 39%] Built target BasicUsageEnvironment
[ 58%] Built target pambase
[ 60%] Built target pamfft
[ 60%] Built target pamlevel
[ 61%] Built target anglemeters
[ 62%] Built target aoipinfo
[ 63%] Built target channeldelay
[ 64%] Built target distortion
[ 67%] Built target fft
[ 68%] Built target fftphase
[ 69%] Built target identify
[ 70%] Built target levels
[ 71%] Built target lineup
[ 73%] Built target lissajou
[ 74%] Built target LTC
[ 76%] Built target LTCGenerator
[ 78%] Built target meters
[ 79%] Built target peakcount
[ 80%] Built target peaklog
[ 81%] Built target polarscope
[ 83%] Built target r128
[ 84%] Built target radar
[ 85%] Built target record
[ 87%] Built target scope
[ 89%] Built target spectogram
[ 90%] Built target InputAlign
[ 91%] Built target waveform
[ 99%] Built target pam2
[100%] Built target pamupdatemanager
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/pam2/libpambase.so
-- Installing: /usr/local/lib/pam2/libpamfft.so
-- Set runtime path of "/usr/local/lib/pam2/libpamfft.so" to ""
-- Installing: /usr/local/lib/pam2/libpamlevel.so
-- Set runtime path of "/usr/local/lib/pam2/libpamlevel.so" to ""
-- Installing: /usr/local/lib/pam2/monitor/libanglemeters.so
-- Set runtime path of "/usr/local/lib/pam2/monitor/libanglemeters.so" to ""
-- Installing: /usr/local/lib/pam2/monitor/libaoipinfo.so
-- Set runtime path of "/usr/local/lib/pam2/monitor/libaoipinfo.so" to ""
-- Installing: /usr/local/lib/pam2/test/libchanneldelay.so
-- Set runtime path of "/usr/local/lib/pam2/test/libchanneldelay.so" to ""
-- Installing: /usr/local/lib/pam2/test/libdistortion.so
-- Set runtime path of "/usr/local/lib/pam2/test/libdistortion.so" to ""
-- Installing: /usr/local/lib/pam2/monitor/libfft.so
-- Set runtime path of "/usr/local/lib/pam2/monitor/libfft.so" to ""
-- Installing: /usr/local/lib/pam2/monitor/libfftphase.so
-- Set runtime path of "/usr/local/lib/pam2/monitor/libfftphase.so" to ""
-- Installing: /usr/local/lib/pam2/monitor/libidentify.so
-- Set runtime path of "/usr/local/lib/pam2/monitor/libidentify.so" to ""
-- Installing: /usr/local/lib/pam2/test/liblevels.so
-- Set runtime path of "/usr/local/lib/pam2/test/liblevels.so" to ""
-- Installing: /usr/local/lib/pam2/test/liblineup.so
-- Set runtime path of "/usr/local/lib/pam2/test/liblineup.so" to ""
-- Installing: /usr/local/lib/pam2/monitor/liblissajou.so
-- Set runtime path of "/usr/local/lib/pam2/monitor/liblissajou.so" to ""
-- Installing: /usr/local/lib/pam2/monitor/libLTC.so
-- Set runtime path of "/usr/local/lib/pam2/monitor/libLTC.so" to ""
-- Installing: /usr/local/lib/pam2/monitor/libLTCGenerator.so
-- Set runtime path of "/usr/local/lib/pam2/monitor/libLTCGenerator.so" to ""
-- Installing: /usr/local/lib/pam2/monitor/libmeters.so
-- Set runtime path of "/usr/local/lib/pam2/monitor/libmeters.so" to ""
-- Installing: /usr/local/lib/pam2/monitor/libpeakcount.so
-- Set runtime path of "/usr/local/lib/pam2/monitor/libpeakcount.so" to ""
[... 100 lines cut ...]
-- Configure plugin polarscope
-- Configure plugin r128
-- Configure plugin radar
-- Configure plugin record
-- Configure plugin scope
-- Configure plugin spectogram
-- Configure plugin InputAlign
-- Configure plugin waveform
-- Configure pam2
-- Checking for one of the modules 'libcurl'
-- Found libcurl
-- Found avahi-client
-- Found portaudio
-- Configure pamupdatemanager
-- Found libcurl
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    optionNMOS

-- Build files have been written to: /home/pi/pam/build

I hope you can find a solution for this problem quickly.

martim01, 2021-12-03:
The CMake build had become a bit of a mess - I think I've fixed it now. I've tested on a vanilla machine and everything now builds without any errors.  I've also automated the adding of the pam2.conf file in to /etc/ld.so.conf.d

closed 2021-12-03 by martim01