martim01/pam #66: Incorrect NMOS installation

Twister10000, 2021-12-14:
Hello Martim

I have noticed the following problem. 
In the CMakeLists.txt file the queries for the NMOS installation are missing.
If you want to install NMOS separately, the following error message always appears during the build:

`10%] Building CXX object /home/pi/nmos/build/CMakeFiles/nmos.dir/src/ioresource.cpp.o
[ 10%] Building CXX object /home/pi/nmos/build/CMakeFiles/nmos.dir/src/is04server.cpp.o
[ 10%] Building CXX object /home/pi/nmos/build/CMakeFiles/nmos.dir/src/is05server.cpp.o
[ 10%] Building CXX object /home/pi/nmos/build/CMakeFiles/nmos.dir/src/log.cpp.o
[ 10%] Building CXX object /home/pi/nmos/build/CMakeFiles/nmos.dir/src/microserver.cpp.o
/home/pi/nmos/src/microserver.cpp: In member function ‘virtual bool MicroServer::Init()’:
/home/pi/nmos/src/microserver.cpp:201:80: error: invalid conversion from ‘int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, size_t*, void**)’ {aka ‘int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, unsigned int*, void**)’} to ‘MHD_AccessHandlerCallback’ {aka ‘MHD_Result (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, unsigned int*, void**)’} [-fpermissive]
  201 |    m_pmhd = MHD_start_daemon (MHD_USE_SELECT_INTERNALLY, m_nPort, NULL, NULL, &MicroServer::AnswerToConnection, this, MHD_OPTION_NOTIFY_COMPLETED, RequestCompleted, NULL, MHD_OPTION_END);
      |                                                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                               |
      |                                                                               int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, size_t*, void**) {aka int (*)(void*, MHD_Connection*, const char*, const char*, const char*, const char*, unsigned int*, void**)}

In file included from /home/pi/nmos/include/server.h:4,
                 from /home/pi/nmos/include/microserver.h:2,
                 from /home/pi/nmos/src/microserver.cpp:1:
/usr/include/microhttpd.h:2443:45: note:   initializing argument 5 of ‘MHD_Daemon* MHD_start_daemon(unsigned int, uint16_t, MHD_AcceptPolicyCallback, void*, MHD_AccessHandlerCallback, void*, ...)’
 2443 |                   MHD_AccessHandlerCallback dh, void *dh_cls,
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~
gmake[2]: *** [/home/pi/nmos/build/CMakeFiles/nmos.dir/build.make:459: /home/pi/nmos/build/CMakeFiles/nmos.dir/src/microserver.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:931: /home/pi/nmos/build/CMakeFiles/nmos.dir/all] Error 2
gmake: *** [Makefile:149: all] Error 2
`

I just used the CMakeLists.txt from the nmos Branch for this error message.

martim01, 2021-12-14:
Hi,
The "nmos" branch of PAM was dependant on the "development" branch of my NMOS library. This morning I merged the development branch back in to the main branch. So the nmos branch of PAM should now build correctly. At some point once I've finished documentation I intend to merge the "nmos" PAM branch in to its main branch.

Twister10000, 2021-12-14:
Hello

Thank you very much for this quick fix.
I will test it today or tomorrow on my system. 
I would then post the feedback in the issue.

martim01, 2021-12-14:
Thanks - the NMOS build is still a bit of a work in progress so any feedback will be very helpful

Twister10000, 2021-12-16:
Hello,

I will get back to you. 
I have now tried it out. 
The logs I am sending are from a system where PAM is already installed so some outputs are strange. 
When cmake ... -DNMOS=ON everything worked fine it fetched all libaries from Github including NMOS.

With cmake --build . everything is also built successfully. 
pi@raspberrypi-PAM:~/pam-nmos/build $ cmake --build .
-- Build Type: Release
-- NMOS: ON
-- Find external libraries
-- Find pml::log
-- log found - update to latest version
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
-- Find pml::ptpmonkey
-- ptpmonkey found - update to latest version
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
-- Find pml::sapserver
-- sapserver found - update to latest version
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
-- Find pml::dnssd
-- pml::dnssd found - update to latest version
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
[... 225 lines cut ...]
-- Up-to-date: /usr/local/lib/pam2/test/liblineup.so
-- Up-to-date: /usr/local/lib/pam2/monitor/liblissajou.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libLTC.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libLTCGenerator.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libmeters.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libpeakcount.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libpeaklog.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libpolarscope.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libptp.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libr128.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libradar.so
-- Up-to-date: /usr/local/lib/pam2/monitor/librecord.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libscope.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libspectogram.so
-- Up-to-date: /usr/local/lib/pam2/test/libInputAlign.so
-- Up-to-date: /usr/local/lib/pam2/monitor/libwaveform.so
-- Up-to-date: /usr/local/bin/pam2

Also, when I try to run pam it always gives the following error message:
pi@raspberrypi-PAM:~/pam-nmos/build $ pam2
pam2: error while loading shared libraries: libnmos_node.so: cannot open shared object file: No such file or directory
pi@raspberrypi-PAM:~/pam-nmos/build $

Can you take a look at this I assume it will be a bug in the CMake config. 
[MobaXterm_pi_20211216_082852.pdf](https://github.com/martim01/pam/files/7725328/MobaXterm_pi_20211216_082852.pdf)

martim01, 2021-12-16:
I'd forgotten to add the install step to the nmos CMake files so they weren't being copied to /usr/local/lib correctly. This has been corrected

Twister10000, 2021-12-16:
It works now

closed 2021-12-16 by Twister10000