martim01/pam #69: Maybe error in NMOS

Twister10000, 2021-12-16:
I found the following when testing NMOS.
On my machine the NMOS port is 8080. 
When I want to access the NMOS API only the x-nmos/ comes up. 
[image]
[image]

martim01, 2021-12-16:
I believe this is correct. You can then do
10.120.8.34:8080/x-nmos which should return ["node/","connection/"]
and then you can navigate to 10.120.8.34:8080/x-nmos/node or 10.120.8.34:8080/x-nmos/connection etc.

Twister10000, 2021-12-16:
You are right if you enter it manually it works fine.

I knew it only that you can click directly on the web page.
Maybe that would be a cool new feature.

closed 2021-12-16 by Twister10000