martim01/pam #71: NMOS Installation

Twister10000, 2021-12-20:
Hello 

When starting pam2, the following error message appears in the terminal. 

**11:01:04.755    INFO    Plugins Monitor Lib: fft
11:01:05.531    INFO    NMOS    Setup NMOS
eth0: 10.120.8.34  dc-a6-32-b1-96-9411:01:05.532        WARNING MongooseServer  AddEndpoint <GET, /x-nmos/node> failed as methodpoint already exists
11:01:05.533    WARNING MongooseServer  AddEndpoint <GET, /x-nmos/connection> failed as methodpoint already exists
eth0: 10.120.8.34  dc-a6-32-b1-96-9411:01:05.535        DEBUG   ++++++++++ Commit source: 7829f488-6e0c-34b7-8f1f-53424ca83e05 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
11:01:05.535    DEBUG   ++++++++++ Commit device: 41082555-4957-39fc-9eca-3cf7458db7c0 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
11:01:05.536    DEBUG   ++++++++++ Commit flow: 6adcfd69-ccb1-33f4-af4e-e3d5549ddea2 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
11:01:05.536    DEBUG   ++++++++++ Commit sender: 5103822b-20fa-35fa-bf4b-4bbc5bb7766c +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
11:01:05.537    DEBUG   ++++++++++ Commit device: 41082555-4957-39fc-9eca-3cf7458db7c0 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
11:01:05.537    DEBUG   ++++++++++ Commit receiver: a44ff54b-507f-33a3-afab-d388ae75e822 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
11:01:05.537    DEBUG   ++++++++++ Commit device: 41082555-4957-39fc-9eca-3cf7458db7c0 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
11:01:05.538    DEBUG   ++++++++++ Commit receiver: a44ff54b-507f-33a3-afab-d388ae75e822 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
11:01:05.538    INFO    NMOS    Start NMOS Services 1
11:01:05.538    INFO    IOManager       Create Audio Output Generator: Generator
11:01:05.538    DEBUG   NMOS: Start Http Servers
11:01:05.539    DEBUG   NMOS: Start Http Servers: Done
11:01:05.539    INFO    NMOS: Start mDNS Publisher
11:01:05.539    DEBUG   pml::dnssd:     10.120.8.34
2021-12-20 10:01:05 2 mongoose.c:3319:mg_mkpipe  pipe 19
11:01:05.539    INFO    IOManager       Generating fixed frequency 1000Hz at -21.4dbFS
11:01:05.539    INFO    Server started: http://0.0.0.0:8080**

I also have problems to configure the Raspberry Pi via NMOS. 
My software always says that under this IP it can not reach anything.

martim01, 2021-12-20:
The WARNING messages are nothing to worry about - they should be changed to TRACE messages.
The IP address 0.0.0.0 refers to the servers listening address - it means the server is "listening to all IPv4 addresses on the local machine". Again not a very useful log message for the user and should be changed to a DEBUG or TRACE message.

Twister10000, 2021-12-20:
THX for the response

closed 2021-12-20 by Twister10000