martim01/pam #83: Error message when starting PAM

Twister10000, 2022-01-26:
Hello 

I am reporting again with a new error.

When starting PAM I always get this error message. 

I have attached a picture of the error message
[image]

martim01, 2022-01-26:
Hi,
I've seen this before - it usually means that one of the libraries that the PAM uses can be found in more than one directory of the path.
Between version 1.2 and 1.3 (I think) I changed the location of the libpam*.so libraries from /usr/local/lib/pam2 to /usr/local/lib

If you've built and installed the later version then it's worth checking that the following
### /usr/local/lib
Has the following libraries
* libnmos_base.so
* libnmos_client.so
* libnmos_node.so
* libpambase.so
* libpamfft.so
* libpamlevel.so
* libpml_dnssd.so
* libpml_log.so
* libptpmonkey.so
* librestgoose.so
* libsapserver.so
### /usr/local/lib/pam2
Has no libraries but 3 directories
* generator
* monitor
* test

If any libraries are in /usr/local/pam2 then try deleting them and restarting.
If this doesn't fix the issue then could you press the "Backtrace" button on the error message and post a screen shot of the result.
Thanks

Matt

Twister10000, 2022-01-26:
Hello

I have checked the directories and everything is now as it should be. 

But now I still get an error message.
I have attached them as images 
[image]
[image]
[image]
[image]
[image]

martim01, 2022-01-27:
The first screenshot still makes me think that you have the same library in two places. There was a brief bug with the install process that put some of the test plugins in to the monitor directory. I wonder if that has happened here.
Check /usr/local/lib/pam2/monitor and check there is nothing in there that is also in /usr/local/lib/pam2/test or .../generator. If there is then delete it from the monitor directory

Twister10000, 2022-01-28:
Hello 
I accidentally deleted a wrong library and could not restore it.
I have therefore rebuilt PAM. With the new build I have no longer this error. 
So I can't tell you if your solution would have solved the problem. What I can say, is that there were indeed certain libraries in both places.

Twister10000, 2022-01-28:
I was able to free the directories from duplicate libraries. 
The error message has now disappeared. 
Thanks a lot for your support with all my questions.

closed 2022-01-28 by Twister10000