martim01/pam #84: Feature Request - Support for 3.0 streams & 5.1 Streams

Twister10000, 2022-02-02:
Hello

I have a feature request.
Would it be possible for you to add the feature to the PAM so that it can also receive 3.0 (Left, Center, Right) and 5.1 streams?

martim01, 2022-02-02:
Do you mean Dolby encoded streams or SMPTE 2210-30 streams with 3/6 audio channels and the relevant channel mapping?
If Dolby then unlikely unless there is an open source decoder library. If 2110-30 then this should work, except currently the channel naming in the SDP is ignored

Twister10000, 2022-02-03:
Hello
Thank you for the response. 
I do not want Dolby. 
My question is that would be possible for you to program an option for 2110-30? It would help me a lot

martim01, 2022-02-06:
Hi,
2110-30 should already be supported up to 8 mono channels. I think only the meters and anglemeter plugins will show the full 8 channels, the others provide a routing option to decide which ones are displayed.

Is this not working for you or, is it the extra channel mapping in the SDP which you would like to see, or plugins being able to analyse more channels at the same time?
Thanks

Twister10000, 2022-02-09:
Hello 

I am sorry that my answer took so long. 
I was able to test it and it works fine:) Thanks a lot. 
I have only one question regarding the R128 plugin. Are all channels taken together there or how does that work exactly? It should be all channels together or?

I would like to ask you if it would be possible to display all 8 chanels in the Scope and Waveform plugin?

martim01, 2022-02-09:
The R128 plugin follows the ITU-R BS.1770  spec. This means that all channels are taken together. However, it currently treats all channels with equal weight when summing them, whilst the spec says that with 5.1 channels the Left and Right surround should have higher weighting. When I get round to adding the decoding of the 211-30 channel-format line of an SDP I'll look at correcting this for channels defined as being surround

I think it should be possible to display all 8 channels in the Scope and Waveform - I'll have a look at how tricky it would be

martim01, 2022-02-11:
Scope and waveform plugin now show up to 8 channels

Twister10000, 2022-02-15:
Thanks the feature works.
Should I leave this issue open because of the adjustments to the R128 or should I close it?

martim01, 2022-02-15:
Hi,
Would you mind closing the issue and opening a new one with the R128 feature. It's more likely that I will remember to look at it if it's a new issue that doesn't have my name on it.
Thanks

closed 2022-02-16 by Twister10000