martim01/pam #94: R128 PPM request

gskips, 2022-03-17:
Hi there Martin,

Thank you or this lovely piece of software. I'm a tv/film editor with an interest in Rpi and this is perfect for my needs, thank you!

I have question though, and I'm by no means a programmer, so sadly not something I could tackle myself, but is it possible to make a plugin modular, so we can chop and change bits of plug-ins with others etc...? Perhaps use the Angle Meters along with a bar-graph meter, or Lissajou.

My main thought would be is it possible to swap out the True Peak meters in R128 and have the BBC PPM meters instead?

So, not really an issue, more a feature request. I know these projects are time-consuming and people coming along asking for this, that and the other isn't helpful - but love using the software, and thanks for the pre-made downloads...so helpful!

(also, for the pre-made packages, what have you set the passwords for pi and pam for 1.3.2... I've managed to make a new sudo user, but the only two passwords I could find on discussions didn't seem to work?)

Cheers
Gary

martim01, 2022-03-17:
Hi Gary,
Thanks for the kind feedback, I'm glad you are finding it useful.
I have occasionaly thought of doing a modular plugin but it's quite a lot of code changing so I've never got round to it. Also I've never quite worked out how to build the designer.
Making it so you can have the normal meters rather than the true peak meters on the R128 plugin is pretty easy though and I'll look at doing that.

In terms of the password it should be qw3rtyu10p
If you look in the /home/pi/Pam folder there is a file called setup. If you delete this and reboot you should get the initial setup where you can enter a sudo password

gskips, 2022-03-18:
Hey Matthew, (so sorry for calling your Martin above!)

Yes, if there was a way of making the True Peak selectable like the Lissajou plugin, that would be amazing. I completely understand how complicated it all is, I'd love to have to coding knowledge but sadly, I'm just an end user in this case... maybe something to work on when our 11 month old boy is a little older ;-) .

Thanks again,

martim01, 2022-04-03:
R128 meter now allows you to have other meter types as well as the true peak

closed 2022-04-03 by martim01

gskips, 2022-04-04:
Hey Matthew, Amazing, thanks very much! - I'll get 1.5.0 as soon as I'm back from hols!

I've always installed with the pre-built image, as building the whole project seems a little daunting for me, is there a way to upgrade from the version I have (1.3.2) to 1.5.0 without starting again with whole system (sorry if this is covered elsewhere, I couldn't find anything after a quick search)? I've only made a few small tweaks to users etc, so no major issue if I have to start again from a fresh SD card, but just wondered if there was an upgrade path to go down instead?

Many thanks again!

martim01, 2022-04-05:
Hi,
There should be an upgrade path, but unfortunately it's currently not working as well as I hoped. It's the next thing on my list to look at. Once it works you'll simply need to plug a USB stick in with the upgrade file and select the update option on settings.

I've done quite a bit of work on building from source however and it really should be as simple as it says on the readme.

Matt

gskips, 2023-01-13:
Hey Matt,

Happy New Year, I hope all is going well for you!?

I just wanted to give you a shout, to ask a quite specific question, coming from someone that dabbles in RPI stuff and really doesn’t know too much beyond searching the internet for answers!!

I’m loving using your PAM, thanks… but I’d love to be able to use the rpi4 for other things too. I currently have a rpi4 set up with 3 virtual VNC desktops running a couple os slideshows and a cctv monitor. - My main question to you, is , is it possible to have PAM running on a Buster rpi4 with the GUI running underneath it, enabling me to run VNC Server as well, with a virtual desktop to log into ?

I guess, ultimately, is it possible for me to start a fresh with Raspian full, and run PAM in a maximised window?? Could this work somehow and maintain all functionality, along with having the option to have the full gun running too?

Sorry if these are obvious, yes's or nos… but I’m no programmer but like to try things and get things working if they can!

Cheers
Gary

martim01, 2023-01-13:
Hi Gary,

Happy New Year to you as well.

It is indeed possible to run the PAM on a RP4 (or any Linux machine) with the GUI underneath - I have it running on an Lubuntu VM on my Windows Laptop for testing stuff.

If you follow the guide in the ReadMe on the GitHub page to build and install PAM then you can easily run it from a Raspberry Pi terminal by typing pam2

Some caveats:
The size of the window is not changeable - it will always be 800x480 pixel
There is no Title bar to the window so if you want to move it you have to press the Alt key, click anywhere on the window and then move the mouse.

Cheers,

Matt

gskips, 2023-01-13:
Hey Matt,

Amazing thanks so much, that’s great news… and the caveats shouldn’t be a problem as I hope to still run pam on a rpi official touch screen, as it is now… but then have the ability to VNC into the gui with virtual desktops, so should work brilliantly! Thanks.

Just one last thing, looking at the readme, looks like a lot of it takes care of itself with the build if using cmake? My question is about cmake. I’ve not used it before, I’m guessing it needs installing on my pi first?

Cheers
Gary

martim01, 2023-01-13:
Yes you probably have to install cmake, I don't think it's on the pi by default.

sudo apt install cmake

should work.

gskips, 2023-01-13:
Ha, amazing thanks, figured it would be a simple install!

… and definitely finally… in the case of an uninstall….? Is everything pretty much contained in the pam directory in home… or are things scattered and not easily removed? (Just building on a test machine, that perhaps I shouldn’t have started!)

Thanks very much again, and thank you for all you help!

Cheers
Gary

martim01, 2023-01-13:
If you just do cmake .. then you'll get a load of other directories in your home if you look at the "Other libraries required" section you should be able to work out the names

If you do the .. target install line then files get
 put in /usr/local/bin and /usr/local/lib

Matt

gskips, 2023-01-13:
Thanks Matt, that’s massively helpful. Will give all this a go soon on a 64bit Raspian install.. hopefully all will work ok!

Cheers
Gary

gskips, 2023-01-15:
Hey Matt,I hope you had a good weekend!Sorry to be bothering you again…I have been having some ‘fun’ attempting to install onto a full version os raspian. First I tried 64bit then 32bit Bullseye… then I tried it on 32bit Buster. Sadly each time it’s failed. I am trying to install your 1.54 ‘latest’ version as this is what I have running on a ‘pre installed’ version that i got from your page.The problem I am having seems to be in the cmake —build . stage. I have attached a copy/paste from the terminal in Raspian from halfway through the unzipping of the downloaded file (this is as far back as it went) - for ease of reading I’ve crudely highlighted the commands I’ve typed, so you can skip through… I’m hoping it makes some sense to you as this is as far as my knowledge goes… and sorry if it’s something obvious that I am or am not doing…. If you have time, I’d really appreciate if you might be able to take a little look and point me in the right direction, if you can please and thank you!Just for info.. all the installs are fresh installs from the pi builder, fully updated. Ideally i wanted it to sit on a 64bit Bullseye, but tried the 32bit versions incase that was causing the problem…. Seems not. I’ve also tried installing as root rather than sudoing, but that didn’t make any difference either.Thanks so much,CheersGary

gskips, 2023-01-15:
Sorry just further to the below. It seems it’s getting to different  stages.  I’m just ran cmake .. again and also —build again and it’s currently 47% and counting. It’s getting a bunch of stuff pop up aside from the “building cxxx” green lines, but I don’t know if that’s normal or not? I’ll see how far it gets now….but it’s really odd that it’s not consistently stopping at the same places??CheersGaryHey Matt,I hope you had a good weekend!Sorry to be bothering you again…I have been having some ‘fun’ attempting to install onto a full version os raspian. First I tried 64bit then 32bit Bullseye… then I tried it on 32bit Buster. Sadly each time it’s failed. I am trying to install your 1.54 ‘latest’ version as this is what I have running on a ‘pre installed’ version that i got from your page.The problem I am having seems to be in the cmake —build . stage. I have attached a copy/paste from the terminal in Raspian from halfway through the unzipping of the downloaded file (this is as far back as it went) - for ease of reading I’ve crudely highlighted the commands I’ve typed, so you can skip through… I’m hoping it makes some sense to you as this is as far as my knowledge goes… and sorry if it’s something obvious that I am or am not doing…. If you have time, I’d really appreciate if you might be able to take a little look and point me in the right direction, if you can please and thank you!Just for info.. all the installs are fresh installs from the pi builder, fully updated. Ideally i wanted it to sit on a 64bit Bullseye, but tried the 32bit versions incase that was causing the problem…. Seems not. I’ve also tried installing as root rather than sudoing, but that didn’t make any difference either.Thanks so much,CheersGary

gskips, 2023-01-15:
Last one from me tonight…. Attached the same, for the second attempt tonight… just for info. Thanks.Sorry just further to the below. It seems it’s getting to different  stages.  I’m just ran cmake .. again and also —build again and it’s currently 47% and counting. It’s getting a bunch of stuff pop up aside from the “building cxxx” green lines, but I don’t know if that’s normal or not? I’ll see how far it gets now….but it’s really odd that it’s not consistently stopping at the same places??CheersGaryHey Matt,I hope you had a good weekend!Sorry to be bothering you again…I have been having some ‘fun’ attempting to install onto a full version os raspian. First I tried 64bit then 32bit Bullseye… then I tried it on 32bit Buster. Sadly each time it’s failed. I am trying to install your 1.54 ‘latest’ version as this is what I have running on a ‘pre installed’ version that i got from your page.The problem I am having seems to be in the cmake —build . stage. I have attached a copy/paste from the terminal in Raspian from halfway through the unzipping of the downloaded file (this is as far back as it went) - for ease of reading I’ve crudely highlighted the commands I’ve typed, so you can skip through… I’m hoping it makes some sense to you as this is as far as my knowledge goes… and sorry if it’s something obvious that I am or am not doing…. If you have time, I’d really appreciate if you might be able to take a little look and point me in the right direction, if you can please and thank you!Just for info.. all the installs are fresh installs from the pi builder, fully updated. Ideally i wanted it to sit on a 64bit Bullseye, but tried the 32bit versions incase that was causing the problem…. Seems not. I’ve also tried installing as root rather than sudoing, but that didn’t make any difference either.Thanks so much,CheersGary

martim01, 2023-01-16:
Hi Gary,

Looks like this is caused by a bug in the CMake code for the Restgoose library. Basically it should clone and then checkout a specific version of the Mongoose webserver library from GitHub but it's only doing that if the library has already been downloaded.

I'll have a look at fixing the issue, but the simple fix is simply to run the initial cmake .. command twice. The first time will clone Mongoose and the second time it will make sure it is the correct version.

I've just tried this on a vanilla Debian VM and it built fine, so hopefully it should also work on a Pi.

If not drop me another email - though I didn't get your attachments for some reason. Maybe try emailing me direct at ***@***.*** rather than via the GitHub email.

Cheers

Matt

gskips, 2023-01-16:
Hi Matt, 

Thanks for getting back to me, and apologies for the bombardment of emails yesterday. I’m very happy to send on the attachments again, but your direct email comes up masked (***@***.***) so not too sure where to send them… any chance yo might be able to drop me a quick mail from that address and I’ll forward them on to you… only because it seems to stop a different points each time?

Cheers
Gary

martim01, 2023-01-16:
Hi Gary – annoyingly your email is just showing as a Github one. My email is my Github username (martim01) and I use outlook.com. Hopefully you can workout my address from that but GitHub won’t **** it