martim01/pam #99: Install Problems Restgoose / SSL

Gemini2350, 2022-08-05:
Hey Martim

I've got 2 Problems with installing:

Without Nmos:
/home/gemini/pam/plugins/lissajou/../../pambase/monitorpluginbuilder.h:9:10: fatal error: RestGoose.h: No such file or directory
    9 | #include "RestGoose.h"

I can solve thisone by Installing with Nmos, but then i got:
[ 90%] Linking CXX executable ../../bin/pam2
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_set_cipher_list'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_set_verify'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_get0_param'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_CTX_load_verify_locations'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `OPENSSL_init_ssl'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `X509_VERIFY_PARAM_set1_host'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_read'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `TLS_server_method'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_connect'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_use_certificate_file'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_set_options'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_use_certificate_chain_file'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_free'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_new'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_set_session_id_context'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_CTX_new'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_set_fd'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_get_error'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_use_PrivateKey_file'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `ERR_print_errors_fp'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_pending'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_ctrl'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_write'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `TLS_client_method'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_CTX_free'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `ERR_clear_error'
/usr/bin/ld: ../../lib/librestgoose.so: undefined reference to `SSL_accept'
collect2: error: ld returned 1 exit status

martim01, 2022-08-05:
**Second issue:** Looks like there's a couple more libraries that need installing currently:
`sudo apt install libssl-dev libcurl4-openssl-dev`
I've added these to the readme
**First issue:** This is an error in the CMake file that still needs fixing

Gemini2350, 2022-08-05:
Thanks. sudo apt install libssl-dev libcurl4-openssl-dev 
solved the second one

closed 2023-07-04 by martim01