martim01/pam #103: Function from sapserver missing

Crowerade, 2023-05-06:
I was trying to build pam with  ` cmake --build . `, and an error appeared killing the process :
```
/home/pi/pam-1.5.4/pambase/aoipsourcemanager.cpp: In member function ‘void AoipSourceManager::StartDiscovery(wxEvtHandler*, const std::set<std::__cxx11::basic_string<char> >&, std::set<std::__cxx11::basic_string<char> >&)’:
/home/pi/pam-1.5.4/pambase/aoipsourcemanager.cpp:425:112: error: no matching function for call to ‘pml::SapServer::SapServer(std::shared_ptr<wxSapHandler>)’
         m_pSapWatcher = std::unique_ptr<pml::SapServer>(new pml::SapServer(std::make_shared<wxSapHandler>(this)));
                                                                                                                ^
```
There seems to be ` sapserver.h ` in the libraries but I found no file with that name.

martim01, 2023-05-09:
Agreed - it looks like the build of release 1.5.4 has become broken over the last year.

I'd suggest you grab the latest commit from the master branch and build from there. I've just tested that and it compiles fine. It should all be as stable as 1.5.4 and has more features. I just haven't had a chance to create a new release recently

dewiweb, 2023-05-23:
Hello, I've got a similar error trying to build from master branch : 
``
/home/pi/pam/pambase/iomanager.cpp:1341:86: error: no matching function for call to 'pml::SapServer::SapServer(std::nullptr_t)'
``

dewiweb, 2023-05-24:
Strange, because I've got this error cloning actual master branch(yesterday) and trying to build from it.

martim01, 2023-05-24:
I updated the SapServer repository yesterday afternoon - so if you tried building before then you would have had the issue.
I think it should be sorted now. You'd need to run ```cmake ..``` again in order to pull down the updated SapServer repository

Crowerade, 2023-06-03:
Seems to work, thanks !
Closing the issue...

closed 2023-06-03 by Crowerade