martim01/pam #104: Alternative vu color for R 128 s2

viktorfreiman, 2023-05-28:
Hi Matthew, I have been loving this tool.

I am working with streaming media and I use R 128 s2 which has -16.0 LUFS as taget level insted of -23.0 LUFS
Currently r128 meters are colored to -23 (green is between -27? to -20).
Is there any way to change to what color is displayed at a specified level?

I have tried to look where the color is set in the code but I have not found it.

It would be super nice to be able to set both the level and what color (I think the blue is a bit to dark for my taste) in a config file to not have the need to rebuild.

https://tech.ebu.ch/docs/r/r128s2.pdf

martim01, 2023-05-29:
This sounds like a good idea. The normal meters are currently defined in a config file so it makes sense to do the same with the R128.

martim01, 2023-06-08:
I've updated the R128 plugin so that it now reads the settings from **r128types.xml**
In this file you can change

- Target level
- Levels to show on the meters
- Range of the meter (min/max)
- The three colours and the levels where the colours change

I've added the R128s2 settings to this file as well as the original ones.

closed 2023-07-04 by martim01