martim01/pam #108: Project does not compile

hackityourselfman, 2023-08-11:
Restgoose is not compiled but fails for mutple errors. First error is:
[image]

martim01, 2023-08-14:
Agreed - this was due to a misconfiguration in a CMake file pulling the wrong Git branch of Restgoose.
I've done some merging and editing and checked that I the project compiles on a vanilla Docker container and on my usual build machine.

closed 2023-08-14 by martim01

hackityourselfman, 2023-08-15:
Thanks for checking it out martim. Did you make this fix available yet?

martim01, 2023-08-15:
Yep - the latest commit should have fixed the issues. I've just done a pull and build from this and it works for me

zabaluyev, 2023-09-29:
Hello. I was trying to compile it today on the Rpi4 with exit with errors on compiling of Restgoose.

martim01, 2023-09-29:
I've just updated the repository and it builds okay on my system.
If you have issues I would try deleting the restgoose directory from your pi and make sure your CMake cache is clean.
There was/is an issue with the incorrect restgoose branch being checked out of GitHub