martim01/pam #114: PAM crashing & cannot select Soundcard as input device

robbiejh12, 2023-12-12:
Hi Matthew,

I'm running PAM 1.7.1 on a Raspberry Pi 4, installed using the pre-built image. After a few weeks using AoIP, I'm now trying to use a soundcard instead. When I select 'soundcard' under 'input devices', PAM appears to restart, reopening back on whichever 'input device' page tab I was on before I pressed 'soundcard'. This means it's impossible for me to choose 'soundcard' as my input device.

The issue occurs regardless of whether the USB sound card I'm trying to use is plugged in. The USB sound card has worked when I've had previous versions of PAM. The issue does not happen with any other input device tab, nor with output devices, where the 'soundcard' tab works as expected and my sound card is recognised. I don't think it's related to the closed issue #89.

Thanks!

mattstegner, 2024-06-01:
I am also experiencing this.
I've tried a Class Compliant USB sound card, and a HiFiBerry DAC+ADC hat.  I am running 1.7.1-7fd9e83, and also reproduced with 1.6.) both from the prebuilt images.
Each Tim eI go to select an input device, when I click on "Settings>Input Device>Audio Monitoring Source>Soundcard..." I very briefly see the list of devices, then PAM will crash and restart itself.

These are from the Pre-built images (found here: https://github.com/martim01/pam/releases)
If there is a specific log I can provide I'd be happy to provide it.

mattstegner, 2024-06-01:
Update: I regretted the prebuilt images back to 1.5.4, and the crash does not happen, so this appears to have been introduced between 1.5.4 and 1.6.1.

martim01, 2024-11-18:
I believe this is caused by a bug in the R128 plugin. You could try removing that from the set of loaded plugins and see if that helps. 
There will be a new image once I've updated the base OS to bookworm

MatthewMartinBBC, 2024-11-26:
New release (1.8.0) which fixes this issue

closed 2024-11-26 by MatthewMartinBBC